gillham/logic_analyzer #19: Capturing Samples

chino, 2015-01-04:
It might be useful to have some description of how to play around with the sampling rate and recording size.  

It's confusing at first when you try something like 10hz and not realize that it's going to take 102s.

Sometimes I seem to notice I have to hit the cancel button even for settings like 4mhz which should return right away.

Also I seem to always get a sample count of 2 even when no samples were collected?

It can be hard as well to notice when a signal is full up/down at times too.

Pins floating due to values next to them can be misleading which is really just general Arduino knowledge but maybe good to note that.

gillham, 2015-01-05:
You should only see the capture hang if you have triggering enabled and it is waiting on a trigger.  Otherwise 4MHz should be snappy.  There could be some hardware version / driver version interaction going on, what Arduino hardware and host hardware/OS are you using?

I'm not sure which sample count is '2' you are referencing.  Are you talking about one of the counters in the '1' debug output?

A brief "how to use and what to expect" section in the README is a good suggestion.  At a bare minimum noting how long a capture should take at the extreme settings is useful.

chino, 2015-01-05:
In my toying around I wasn't using triggers at all.

I'm not sure what was wrong though but at times 4hmz definitely got stuck.

I tried using Uno and Duelmilanueve with an OSX system.

When a capture is done I click on "Acquisition Details" and noticed it would capture 2 samples at minimum even when nothing was really captured at all.

gillham, 2015-02-07:
I added some diagnostic commands via issue #18 and it should resolve this issue as well.  The OLS client actually calculates the length of the capture based on the settings, so I'm not sure if it is useful to make my own table.
I will be adding some documentation on the wiki with a basic example of how to use it at some point.

closed 2015-02-07 by gillham