gillham/logic_analyzer #20: Continuous Capture Mode?

chino, 2015-01-04:
Do you think it would be possible to listen/stream data into the UI in real time?

chino, 2015-01-05:
Hm, if the client did support it though it would be pretty cool.

I mean right now if you want to capture at 4mhz you only get a very tiny window to capture data which I would think makes it very hard to use?

chino, 2015-01-05:
Have to wonder how hard it really is to modify the UI to support a continuous rendering mode or as suggested in https://github.com/gillham/logic_analyzer/issues/18 perhaps an ascii-art/ncurses rendering would be an awesome way to add a continuous rendering in the console..

gillham, 2015-01-05:
The 4MHz rate is useful for sampling a continuous signal and you can use a trigger to initiate the capture.  It is probably useful to wrap the DEBUG stuff with an '#if 0' to disable it on the 4MHz capture so it captures immediately after the trigger fires.   You're much better off with a real OLS though if you need to sample that fast.

You can continuously sample with text output with a fairly generic sketch that does something like this in loop():
Serial.print(CHANPIN, BIN);
Or
Serial.println(CHANPIN, BIN);

That will spew as fast as it can over 115200 baud forever which would get you about 1,152 samples per second max.  (115200 / 10 bits = 11520 characters / 8 binary bits + CR/LF = 1152 samples printed per second)
If you just send the byte as binary (as opposed to printing it as an 8 digit string) you can do 11,520 samples per second.  Still pretty slow and no client support for this without manually dumping to a file and then trying to load the file into the UI.

Ask Jawi if he has plans to add any real-time / continuous mode and what hardware supports it.  I don't think it is useful at this point and I would suggest a Saleae Logic 4 ($99) or Logic 8 ($199) if you really need continuous sampling.  The Logic 4 and Logic 8 support analog in addition to digital.

chino, 2015-01-05:
You could probably do something like `cat $device | rube -e 'loop { puts STDIN.read(1).unpack("b8") }'` or similar.

Cool thanks for the advice I'm new to all of this.

Saleae looks really nice!

I've also noticed this: http://logicsniffer.gadgetfactory.net/

I just wanted to play around with IR to learn and didn't want to waste a bunch of money.

chino, 2015-01-05:
Hm any reason I couldn't modify the logic-analyzer to support the analog pins as well?

chino, 2015-01-06:
I guess perhaps it's `ols` it self which doesn't support rendering analog data?

gillham, 2015-01-06:
Right, the LogicSniffer (OLS client) does digital logic only, no oscilloscope (analog) stuff.

chino, 2015-01-06:
I opened a ticket in `ols` about continuous capture and I guess the first answer makes sense: https://github.com/jawi/ols/issues/218

On that note too I'm playing around with some custom sketches right now and will probably try this out for curiosity sake: https://github.com/konsumer/arduinoscope

Harvie, 2019-06-30:
RLE merged in #8 might help a lot...