gillham/logic_analyzer #22: Unrolled loops?

chino, 2015-01-05:
I'm very curious about this.

I understand the concept of "unrolled loops" but why couldn't the compiler have done it for you?

For example here:

https://github.com/gillham/logic_analyzer/blob/master/logic_analyzer_inline_2mhz.ino#L93

gillham, 2015-01-05:
The gcc compiler bundled with the Arduino IDE ignores the unroll-loops options.
For example:
#pragma GCC optimize ("unroll-loops") generates "warning: ignoring #pragma GCC optimize"
__attribute__((optimize("unroll-loops"))) generates "warning: 'optimize' attribute directive ignored"

So as far as I know there is no way to convince it to unroll a specific loop.

chino, 2015-01-06:
Hm.  The optimization steps don't do it implicitly either?

Does it really matter to just use a normal loop?  I guess that adds another jump instruction?

Would be cool to actually have a pre-build step that uses a script to generate the unrolled c code.

chino, 2015-01-06:
It's interesting to think about.  Would a `jump` really matter with the avr architecture?  Like does it really eat up a cycle..

chino, 2015-01-06:
I guess no matter what though if it can't unroll by it self the the test condition would eat up cycles.

closed 2015-01-06 by chino

gillham, 2015-01-06:
Right, all instructions take cycles and a loop needs a check on the counter and then a branch to loop or exit.  That check overhead eats cycles which reduces the sample rate.