gillham/logic_analyzer #26: Problems on Arduino Mega

mkeller0815, 2015-06-27:
I've tried to get the code working on an Arduino Mega 2560. 
The code runs fine under a Arduino Uno, but if I upload the code on a Mega 2560 it does not react on any serial command. Even with DEBUG enabled, I get no answer for an "?" sent over the serial line. 
I'm not sure what is broken, because if I add a 
```
 while(1) { 
   blinkled();
   Serial.println("Test");
 }
```
at the and of  the setup() function the LED blinks, but the String is not written to the Serial connection. 
Doing the same in a minimal sketch on the same Mega board, the text is written to the serial connection.
```
#define ledPin 13
void setup() {
  Serial.begin(115200);
  pinMode(ledPin, OUTPUT);
  while(1) { 
    blinkled();
    Serial.println("Test");
 }
}

void loop() {
}

void blinkled() {
  digitalWrite(ledPin, HIGH);
  delay(200);
  digitalWrite(ledPin, LOW);
  delay(200);
}
```
Just to be sure I also removed the timer2 pin toggle from the code, because it does not work on the Mega, but without any effect.

Mario.

stickbreaker, 2015-06-27:
I did not have a problem, when I used my Mega2560 last week. I just copy all 3 sketches into a directory. I had to add a define for byte, (using Arduino 1.6.5).  It worked fine with  OLS <http://ols.lxtreme.nl>

mkeller0815, 2015-06-28:
OK, it seems to be a problem with the Arduino IDE 1.0.5 that was stil running on my Mac. I tested with the current version 1.6.5 and it seems to work now. The OLS client can now detect the Arduino Mega. 
I'll close this issue now.
Here's my working configuration:
Mac OS X 10.9.5
Java(TM) SE Runtime Environment (build 1.7.0_67-b01)
Arduino IDE 1.6.5 
LogicAnalyzer version: Release: v0.13 February 7, 2015
OLS version 0.9.7.2

closed 2015-06-28 by mkeller0815