gillham/logic_analyzer #31: nice code, great work, thanks

dhiltonp, 2016-02-04:
Thanks, @gillham for your great work on this project.

I feel like it's been fairly straightforward to jump into and contribute, which isn't always the case :). Thanks for putting in so much effort to make this project; I know how much time it takes to try and get everything working just right!

EUA, 2020-02-03:
Yeah, I also want to thank you @gillham 
Just use your project as template for create [similar tool](https://github.com/EUA/ESP32_LogicAnalyzer) for ESP32 devices.

gillham, 2021-12-21:
Thanks for the comments.   
@EUA I'm glad you made a SUMP LA for ESP32 but you'll notice my files use a 2 clause BSD license and you removed that and added the GPL to yours.  I personally don't think that is correct.

closed 2021-12-21 by gillham