gillham/logic_analyzer #32: restrict to 4 channels, get 2x buffer

dhiltonp, 2016-02-04:
What do you think about restricting input to 4 channels? It would then be easy to double-up on memory, giving 2x the buffer space.

markkimsal, 2016-09-10:
:+1: 
I don't personally need to sniff more than 4 wires at a time.  uart spi twi canbus all use 4 or fewer, right?

gillham, 2019-05-15:
I don’t know that the SUMP metadata spec allows the buffer size to change based on the sample rate. Possibly. 
This would significantly complicate the code that is hand padded with nop instructions to get the timing as precise as possible. 
I’ve been considering splitting the code into separate sketches for different hardware that isn’t as easily supported with #ifdef now. That might make it easier to do special versions with fewer lines and more buffer at the cost of lower sample rates as well. 
Thoughts?