gillham/logic_analyzer #33: It indicated that all CH340 based arduino board fail be detected.

ulion, 2016-02-29:
I tried all my arduino boards, all of them failed, except one which is not using CH340 as usb-ttl module.

There are a lot cheap arduino boards using CH340. when I use them, it detect metadata got null.

Maybe related to auto-reset, but I tried 120 ohm resistor and also 100 uF capacitor, they may make the board not auto-reset, but still can not let ols detect the arduino device.

tomaskovacik, 2016-09-28:
ch340 cant make 115200 bps :/

gillham, 2019-05-15:
I have a couple cheap boards with CH340 to test more.  I’m hoping to pick my Arduino stuff up again after moving and a lot of other changes.

gillham, 2021-12-21:
I have tested several CH340 based boards.  Uno style clones, Nano clones etc.  The all generally work at 115200.

@tomaskovacik are you still having problems?  Is there a specifc board on eBay or AliExpress that you have tried that doesn't work?  Send an item number.

tomaskovacik, 2021-12-21:
hello, I do not use your project anymore,  so I can tell what board it was which did not work for me