gillham/logic_analyzer #34: Detection failed! null on LogicSniffer v0.9.7.2

Sekhmet, 2016-04-10:
Hi,
I just love this project and wanted to try this, but sadly it doesn't work for me. I think I'm doing something wrong. I've compiled logic_analyzer (loaded all 3 files, compiled them succesfuly and loaded on my board), then I installed LogicSniffer v0.9.7.2, and wanted to start capture - Connection type serial port, port selected, speed 115200bps and device type Arduino Generic Logic Analyzer, but it doesn't capture anything. When I click on "Show device metadata" it prints "Detection failed! null"
When I click on this test button both TX/RX LEDs blink on my board.

I'm using RobotDyn NANO Arduino with ATmega328, LogicSniffer v0.9.7.2 and Windows 10.

Any ideas what I'm doing wrong?

Thanks.

dhiltonp, 2016-04-11:
It sounds like you're pointing using the wrong port.

Sekhmet, 2016-04-11:
Thanks,

I recompiled it few times and suddently capturing works, but detection has
same problem (it doesn't detect device in capture window)/

gillham, 2021-12-21:
Please re-open if this is still a problem.

closed 2021-12-21 by gillham