gillham/logic_analyzer #39: cannot upload the compiled binary

eadmaster, 2017-10-12:
I've compiled the sketch with the arduino libraries in my debian jessie setup, but it fails to upload (tried with 2 different boards):
```
...
make[1]: Entering directory '/tmp/r/logic_analyzer-master/logic_analyzer-master'
/usr/share/arduino/hardware/tools/avr/../avrdude -q -V -D -p atmega328p -C /usr/share/arduino/hardware/tools/avr/../avrdude.conf -c arduino -b 57600 -P /dev/ttyACM1 \
                -U flash:w:build-uno/logic_analyzer.hex:i
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
```

I've also tried uploading the sketch in the more recent IDE v1.8 but it still fails....

imbuedHope, 2017-10-14:
This is a hardware/bootloader issue that isn't related to the code in this project. It is usually caused by a poor USB connection to the Arduino or the firmware not jumping to the bootloader due to your old code that is still using the serial port. Check the wire and try hitting reset right as you start to flash the board.

You can avoid this entirely if you pick up an ISP Programmer for Aduinos and flash the chip directly.

eadmaster, 2017-10-27:
I've finally figured it out, i had to change the baudrate from 57600 to 115200 in the makefile.
`[DETECTED]           MONITOR_BAUDRATE = 115200  (in sketch)`

closed 2017-10-27 by eadmaster