gillham/logic_analyzer #40: question , is it possible to run it on arduino nano( 328) or arduino uno ?

gnkarn, 2018-01-26:
it compiles ok, but it seams its not doing anything , i get no single output on the serial port. tis

dhiltonp, 2018-01-26:
Try reducing the capture size at logic_analyzer.ino:164 (https://github.com/gillham/logic_analyzer/blob/master/logic_analyzer.ino#L164)

dhiltonp, 2018-01-26:
If that doesn't help, maybe your triggers aren't set up correctly?

gnkarn, 2018-01-26:
Thanks for your answer , in my case as it is a nano , those defines are skipped then size should already be. 532 as per the “else”. 
Am I right ¿

gnkarn, 2018-01-26:
Triggers do you mean on the client application ¿ I have not selected any trigger , just left that option unchecked.

dhiltonp, 2018-01-26:
Without a trigger, you will see no data.

logic_analyzer doesn't transfer data continuously. You set a trigger, when
the trigger is tripped and gathered, data is transferred to your client.

David

gnkarn, 2018-01-26:
enabled  the simple trigger , tested with different channels , but it seams it is not working on arduino nano 328.

gnkarn, 2018-01-26:
it works with arduino uno , but i couldn make it decode or detect a given bit sequence , i think that if you have a standard protocol  that is on the sniffer client  list , then it will be a great tool, but if you are trying to detect a non standard pattern ( in my case a given 16 bits word , where each bit is encoded with mark space ) , i will see if can found other more flexible sniffer .

closed 2018-01-26 by gnkarn