gillham/logic_analyzer #43: build error

abdelrahma22, 2018-03-18:
Arduino: 1.6.13 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Users\EssaM\AppData\Local\Temp\cclwLT9s.ltrans0.ltrans.o: In function `main':

cclwLT9s.ltrans0.o:(.text.startup+0x210): undefined reference to `captureInline4mhz()'

cclwLT9s.ltrans0.o:(.text.startup+0x21e): undefined reference to `captureInline2mhz()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

stickbreaker, 2018-03-18:
@abdelrahma22 
copy all of the sketches into one directory.

abdelrahma22, 2018-03-18:
i had but no good

abdelrahma22, 2018-03-18:
can you give me binary file for ardyuno uno

stickbreaker, 2018-03-18:
[Logic_Analyzer.ino.standard.zip](https://github.com/gillham/logic_analyzer/files/1823274/Logic_Analyzer.ino.standard.zip)

abdelrahma22, 2018-03-18:
it compiled successfully but i cant run the analyzer please help me

stickbreaker, 2018-03-19:
upload the sketch to the UNO

What Serial port is the UNO on?

have you downloaded the OLS client?

Download [ols client](https://www.lxtreme.nl/ols/ols-0.9.7.2-full.zip), extract the entire zip file

navigate to where you extracted the zip file

double click on run.bat

in OLS
  Capture>Device>OpenBench Logic Sniffer
  Capture>Begin Capture - Connection
    Connection Type > Serial Port
    Analyzer Port > UNO PORT from above
    PortSpeed> 115200
    DeviceType> Arduino Generic LogicAnalyzer
    Show Device MetaData>
      LEDs should blink on Uno,  
          My info: DeviceType: AGLAv0
          Firmware: 0.13
          Protocol 2
 Capture>Begin Capture - Triggers
   Uncheck trigger enabled:  until you need them
 Capture>Begin Capture - Acquisition
   sampling Rate: select one
   Record size: what ever, with a uno 1k is max

 Capture>Begin Capture 
   click "Capture"

abdelrahma22, 2018-03-19:
i had 5 channel how i connect them in arduino to the project

stickbreaker, 2018-03-19:
connect A0 .. A5

stickbreaker, 2018-03-19:
This SUMP protocol compatible logic analyzer for the Arduino board supports
5 channels consisting of digital pins 8-12, which are the first 5 bits (0-4)
of PORTB. Arduino pin 13 / bit 5 is the Arduino LED, bits 6 & 7 are the
crystal oscillator pins.

D8.. D13

abdelrahma22, 2018-03-19:
which the pin is analog input

stickbreaker, 2018-03-19:
a logic analyzer is a digital representation of  the signal either on or off, this system cannot display an analog signal.

abdelrahma22, 2018-03-20:
how can i use trigger to start capturing with start condition

Escalion, 2018-04-30:
@abdelrahma22 - Read the README. You need to use a SUMP client and set the trigger there.

gillham, 2019-05-15:
I'll add some verbiage to the README to make sure people are properly creating the 'logic_analyzer' directory in their sketchbook and putting the three files in there.  Apparently not everyone knows how to use 'git clone' to get things off github, so I need to improve the docs. :)

closed 2019-05-15 by gillham