gillham/logic_analyzer #45: [Arduino IDE 1.8]Sketch can't compile

ManfredEP, 2019-05-14:
Since this is actual and all others are more than one year old this is a new one: 

The sketch is not able to compile due to variable problems. One variable is used but not declared. This is basic. It means it can never compile any more. `Arduino: 1.8.5 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Programme\Arduino\arduino-1.8.3\arduino-builder -dump-prefs -logger=machine -hardware C:\Programme\Arduino\arduino-1.8.3\hardware -hardware C:\Users\Admin\AppData\Local\Arduino15\packages -tools C:\Programme\Arduino\arduino-1.8.3\tools-builder -tools C:\Programme\Arduino\arduino-1.8.3\hardware\tools\avr -tools C:\Users\Admin\AppData\Local\Arduino15\packages -built-in-libraries C:\Programme\Arduino\arduino-1.8.3\libraries -libraries C:\Users\Admin\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10805 -build-path C:\Users\Admin\AppData\Local\Temp\arduino_build_57153 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -prefs=runtime.tools.avrdude.path=C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.arduinoOTA.path=C:\Programme\Arduino\arduino-1.8.3\hardware\tools\avr -verbose E:\Program Files\logic_analyzer-master\logic_analyzer_inline_4mhz\logic_analyzer_inline_4mhz.ino
C:\Programme\Arduino\arduino-1.8.3\arduino-builder -compile -logger=machine -hardware C:\Programme\Arduino\arduino-1.8.3\hardware -hardware C:\Users\Admin\AppData\Local\Arduino15\packages -tools C:\Programme\Arduino\arduino-1.8.3\tools-builder -tools C:\Programme\Arduino\arduino-1.8.3\hardware\tools\avr -tools C:\Users\Admin\AppData\Local\Arduino15\packages -built-in-libraries C:\Programme\Arduino\arduino-1.8.3\libraries -libraries C:\Users\Admin\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10805 -build-path C:\Users\Admin\AppData\Local\Temp\arduino_build_57153 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -prefs=runtime.tools.avrdude.path=C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.arduinoOTA.path=C:\Programme\Arduino\arduino-1.8.3\hardware\tools\avr -verbose E:\Program Files\logic_analyzer-master\logic_analyzer_inline_4mhz\logic_analyzer_inline_4mhz.ino
Using board 'mega' from platform in folder: C:\Programme\Arduino\arduino-1.8.3\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Programme\Arduino\arduino-1.8.3\hardware\arduino\avr
Detecting libraries used...
"C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Programme\Arduino\arduino-1.8.3\hardware\arduino\avr\cores\arduino" "-IC:\Programme\Arduino\arduino-1.8.3\hardware\arduino\avr\variants\mega" "C:\Users\Admin\AppData\Local\Temp\arduino_build_57153\sketch\logic_analyzer_inline_4mhz.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Programme\Arduino\arduino-1.8.3\hardware\arduino\avr\cores\arduino" "-IC:\Programme\Arduino\arduino-1.8.3\hardware\arduino\avr\variants\mega" "C:\Users\Admin\AppData\Local\Temp\arduino_build_57153\sketch\logic_analyzer_inline_4mhz.ino.cpp" -o "C:\Users\Admin\AppData\Local\Temp\arduino_build_57153\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Programme\Arduino\arduino-1.8.3\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Admin\AppData\Local\Temp\arduino_build_57153\preproc\ctags_target_for_gcc_minus_e.cpp"
Sketch wird kompiliert...
"C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os  -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Programme\Arduino\arduino-1.8.3\hardware\arduino\avr\cores\arduino" "-IC:\Programme\Arduino\arduino-1.8.3\hardware\arduino\avr\variants\mega" "C:\Users\Admin\AppData\Local\Temp\arduino_build_57153\sketch\logic_analyzer_inline_4mhz.ino.cpp" -o "C:\Users\Admin\AppData\Local\Temp\arduino_build_57153\sketch\logic_analyzer_inline_4mhz.ino.cpp.o"
E:\Program Files\logic_analyzer-master\logic_analyzer_inline_4mhz\logic_analyzer_inline_4mhz.ino: In function 'void captureInline4mhz()':

logic_analyzer_inline_4mhz:54: error: 'trigger' was not declared in this scope

   if (trigger) {

       ^

logic_analyzer_inline_4mhz:55: error: 'trigger_values' was not declared in this scope

     while ((trigger_values ^ CHANPIN) & trigger);

             ^

logic_analyzer_inline_4mhz:55: error: 'CHANPIN' was not declared in this scope

     while ((trigger_values ^ CHANPIN) & trigger);

                              ^

logic_analyzer_inline_4mhz:71: error: 'DEBUG_ENABLE' was not declared in this scope

   DEBUG_ENABLE;

   ^

logic_analyzer_inline_4mhz:72: error: 'DEBUG_ON' was not declared in this scope

   DEBUG_ON;

   ^

logic_analyzer_inline_4mhz:74: error: 'DEBUG_OFF' was not declared in this scope

   DEBUG_OFF;

   ^

logic_analyzer_inline_4mhz:93: error: 'logicdata' was not declared in this scope

   logicdata[0] = CHANPIN;

   ^

logic_analyzer_inline_4mhz:93: error: 'CHANPIN' was not declared in this scope

   logicdata[0] = CHANPIN;

                  ^

logic_analyzer_inline_4mhz:14444: error: 'readCount' was not declared in this scope

   for (i = 0 ; i < readCount; i++) {

                    ^

exit status 1
'trigger' was not declared in this scope`
The error messages of ide using this sketch. since I'm not able to check it in all details now the process is stopped. Possibly later. Now with another one if any.

gillham, 2019-05-15:
Are you sure you cloned the repository correctly?  You should have something like 'Documents/Arduino/logic_analyzer' directory with these files in it:
Makefile
README
logic_analyzer.ino
logic_analyzer_inline_2mhz.ino
logic_analyzer_inline_4mhz.ino
ols.profile-agla.cfg
ols.profile-aglam.cfg

It appears your IDE isn't trying to compile the sketch properly.   When you open the sketch it should open 'logic_analyzer.ino' and the other two files will be tabs in the IDE.  I'll test it on my Windows 10 laptop shortly, but it works fine on a Mac with Arduino 1.8.9.

ManfredEP, 2019-05-15:
Hello,
you're wright! I have'nt used it properly. Due to the kind of handling such files in windows the ide have made three direwctories with the '.ino'-Files, one per dir. Of course that doesn't work. Please excuse me. Perhaps it makes me a little bit less foolish to see that I'm german and the english description of the sketch makes me believe there are three different sketches, one up to 1MHz, second from there to 2MHz and than for the rest until 4 MHz. So the error was concreted. 

So the error message can be closed since it was only a jackass making something wrong. 

greetings

Manni

closed 2019-05-15 by ManfredEP