gillham/logic_analyzer #46: Please make this installable through arduino library manager

Harvie, 2019-06-29:
If you add file `library.properties`, you can then request this for adition in arduino library manager here: https://github.com/arduino/arduino/issues
This will allow people to download and install this to their arduino using few simple clicks.

gillham, 2021-12-20:
This isn't really meant to be a library, it is a standalone application.  Does that make sense?

Harvie, 2021-12-22:
Yes it does. Some projects are distributed this way already. You install "library", then go to "examples" in arduino menu and install the application. It's bit more user friendly than having to download package and manualy copy it to arduino IDE. For example the GRBL is installed in similar manner.

gillham, 2023-10-06:
Ok, I've rearranged the repository to support installing via the 'Add .ZIP Library' in the Arduino IDE.  Tested with Arduino IDE 2.2.1 on Linux.

Added to the repositories.txt via https://github.com/arduino/library-registry/pull/3432

gillham, 2023-10-06:
This is now available in the Arduino IDE.  Search for 'LogicAnalyzer'

closed 2023-10-06 by gillham