gillham/logic_analyzer #49: Analog input

Harvie, 2019-06-30:
Atmega328 can sample analog data using ADC at 10kHz, i think we can provide analog sampling at low samplerates. Something between 1 to 10kHz seems doable (depending on number of analog pins used)...

Not sure if SUMP protocol allows this, but sigrok/pulseview does, so we might add some custom SUMP extension (without breaking compatibility with legacy SUMP software).

gillham, 2021-12-20:
Are you talking about sampling similar to the current logic analyzer where we sample 1024 bytes of data and stop?  Or are you talking about continuously reading samples and reporting via USB serial?  Also how many pins would be sampled?  The issue is each sample takes a byte of data so you could sample 1 pin 1024 times, or 6 pins ~170 times.  
There are a few modes to support.  1 channel x 1024 samples, N x 1024/N samples, continuous streaming samples (115200/10 = 1440 bytes/s max) etc.

With continuous sampling you could do roughly 1KHz I guess?

Anyway, I'm not sure it would be a good feature set for the logic analyzer code, but could be a separate firmware image.

What would be a typical use case?

Harvie, 2021-12-22:
Use case would be to have simple oscilloscope-ish device.
I really like using logic analyzers with sigrok/pulseview software. and pulseview already has some basic support for analog samples. Eg.:

Digital:
[image]

Analog:
[image]