gillham/logic_analyzer #53: Divergency between values

FernandoGarcia, 2022-02-24:
Hi!

Sorry if I'm bothering you.

I'm trying to decode a protocol and I'm using an Arduino Mega running your code as logic analyzer.

The problem is I'm getting divergent values between your logic analyzer code and Arduino `pulseIn `function or even using interrupts.

For this reason I'm not able to decode the protocol.

Could you please show me in a simple example what's the method used to measure pulse width in your code?

Looking to your code I wasn't able to understand.

Here the values I'm getting using interrupts:

[image]

Here the code:

```
#if defined(ESP8266)
void ICACHE_RAM_ATTR falling();
void ICACHE_RAM_ATTR rising();
#define PIN D8  // PIN where the PWM singal arrives
#else
#define PIN 2  // PIN where the PWM singal arrives
#endif

// global variables
volatile unsigned int pulse_width = 0; 
volatile unsigned int prev_time = 0; 
 
void rising() {
  attachInterrupt(digitalPinToInterrupt(PIN), &falling, FALLING);  // when PIN goes LOW, call falling()
  prev_time = micros();
}
 
void falling() {
  attachInterrupt(digitalPinToInterrupt(PIN), &rising, RISING);  // when PIN goes HIGH, call rising()
  pulse_width = micros() - prev_time;
  Serial.println(pulse_width);
}
 
void setup() {
  pinMode(PIN, INPUT_PULLUP);

  Serial.begin(115200);
  attachInterrupt(digitalPinToInterrupt(PIN), &rising, RISING);  // when PIN goes HIGH, call rising()
  sei();  // enable interrupts
}
 
void loop() { }
```
I'm getting similar results using `pulseIn` function.

Thanks in advance.

tmk, 2022-02-24:
You should use OLS instead of pulseview.
https://github.com/gillham/logic_analyzer#client-software

I believe this doesn't support sigrok pulseview yet,  see this issue.
https://github.com/gillham/logic_analyzer/issues/38

FernandoGarcia, 2022-02-27:
Hi @tmk!

Thanks for point the right way.

Now I have managed to decode the protocol of a parking sensor.

[image]

I'm so happy!

@gillham, thanks for your great job!

Best regards.

closed 2022-02-27 by FernandoGarcia