gillham/logic_analyzer #55: Add supported chips to readme

anakimluke, 2023-10-07:
I'm finding it hard to figure what chips these sketches are compatible with.

I see a mention of the  ATmega168, ATmega328 and ATmega2560 chips on the readme, but that's about it.

I also found the `architectures=avr` line on library.properties, but does that mean it is supported on every AVR chip? Or are there exceptions?

Btw, thanks for your work!! Very cool ;)

gillham, 2023-10-10:
Great suggestion, I'll update the README.  Back in 2011 it was a lot more obvious which hardware was supported as there wasn't much else.  That has definitely changed a lot!

I'll add the specific board models I can based on what I can find in the documentation, but I won't be able to make a truly exhaustive list.  Hopefully enough to get people going.

gillham, 2023-10-10:
Fixed it.  Thanks again for the suggestion.

closed 2023-10-10 by gillham