gillham/logic_analyzer #56: Error flushing serial port on Windows

Sonic-Amiga, 2023-10-18:
Capture doesn't seem to work. After i click "run" the button changes to "Stop" for a moment, then quickly changes to "run" again. sigrok log shows:

serial-libsp: Error flushing port (995): �������� �����/������ ���� �������� ��-�� ���������� ������ ������ ��� �� ������� ����������..
pv: Acquisition took 0.04 s

(error description is provided by the OS in russian, it's ok that it cannot be read)

The board is a local Nano clone, using CP210x .

Sonic-Amiga, 2023-10-19:
Update: that was from last stable version from the download page. I downloaded nightly, and "Error flushing" is gone, but still doesn't work. It throws error 995 while reading data. I. e. capture starts, works for a moment, then stops with this error.

Sonic-Amiga, 2023-10-20:
After some more checking, it's not your bug, it belongs to sigrok's Windows serial library. It's exactly https://www.sigrok.org/bugzilla/show_bug.cgi?id=1830

gillham, 2023-10-23:
Thanks for the update.  Would you be able to provide a link to a Arduino Nano clone with the CP2102 (or whatever you have) that is similar to what you are using?  I need to acquire a couple more test units, with CP2102 and CH340 (or similar).

Sonic-Amiga, 2023-10-23:
If you're looking at buying this particular board; sorry, it's made locally here (at least officially), on other side of the New Iron Curtain. Here's a schematic FYI; but really nothing special. https://static.chipdip.ru/lib/724/DOC002724132.pdf

Sonic-Amiga, 2023-10-23:
> If you're looking at buying this particular board; sorry, it's made locally here (at least officially), on other side of the New Iron Curtain. Here's a schematic FYI; but really nothing special. https://static.chipdip.ru/lib/724/DOC002724132.pdf

I guess you can put it together with ease out of any CP2102 eval board